aredn/aredn #14: New Device: GL-AR750S-EXT "SLATE"

k1ky, 2020-12-16:
Request to Add this new "Gigabit" model or review compatibility with existing code to support this device.  Featuring Gigabit Ethernet ports.
We have a couple of these devices on hand and available for testing as needed.

dman776, 2022-04-27:
consider for Openwrt 21.02.x support

stuieordie, 2022-05-13:
Has the GL-AR750S-EXT been added yet? I'm tempted to try the AR-750 firmware.

ae6xe, 2022-05-13:
The ar750 creta is supported inow.  the ar750 slate is not yet available.  We have to upgrade to current openwrt to inherit this support.  I have early images for the slate I've been trying to move forward, but still some critical functionality not yet working.

aanon4, 2023-02-08:
There is a nightly build for this device now, but it's untested as we dont have hardware. If anyone wants to give it a try and report back please do.

k1ky, 2023-02-12:
Can you point us to the "correct" nightly for this unit to test?  There appear to be several that could be confusing.

aanon4, 2023-02-12:
The unhelpful answer is that I don't know. I dont have one of these so I've no way to find out.

k1ky, 2023-02-21:
See issues #729 and #730 
Closing this issue in favor of them.

closed 2023-02-21 by k1ky