aredn/aredn #16: Add additional 2.4Ghz channels between 2.3 and 2.4GHz within Euro Amateur Radio allocation

jpronans, 2020-12-16:
Further to Eoghan's post in https://www.arednmesh.org/content/24ghz-band. This is just a ticket to capture the request to extend the lower end of the 2.4GHZ to be dropped below its current value to make extra channels available to stations in Europe. 

Regards
John
EI7IG

dman776, 2022-07-13:
Is this resolved with ch -3 and -4?

ae6xe, 2022-07-14:
ch -3 and -4 are in the current release and have been tested successfully.   IMO, this can be closed.  However, we are seeing an issue where a known -2 usage, is also reporting a fake ch -3 or -4 signal with iw info commands.  This may or may not be related, but would be separately tracked as a defect/issue.

closed 2022-07-21 by ae6xe