aredn/aredn #17: Enhancement Request: Display a text on node homepage when newer firmware version is available

idecmesh, 2020-12-16:
This message is an enhancement request for the homepage of each node, to display a message when a new stable and/or nightly firmware newer than the one running is available.

As a suggestion on how this could be done: 

Create a RES API URL returning the latest stable versions, and latest nightly build versions.
For example https://www.arednmesh.org/api/latest_firmware/ , returning something like:
{
    "ubnt-bullet-m": [ { "stable": "3.30.4", "nightly": "1105" } ],
    "ubnt-nano-m":   [ { "stable": "3.30.4", "nightly": "1105" } ],
    "ubnt-rocket-m": [ { "stable": "3.30.7", "nightly": "1107" } ],
    ...etc...
}

This can then be queried by code on the node homepage, compared against current version, and
* if current_version < latest_stable then display "new stable version available !"
* if current version < latest_nightly then display "new nightly version available !"

I am sure this is very simplistic and might require significant changes in the versioning mechanism, just wanted to plant a seed for future revisions.

Will be happy to help in any way I can.

Thanks !

cTurtle98, 2022-12-31:
@ae6xe I think release 3.22.8 closes this issue?

k1ky, 2024-09-04:
Isn't this taken care of in the new UI?

aanon4, 2024-09-04:
It is.

closed 2024-09-04 by aanon4