aredn/aredn #19: DTD SWCONFIG Options by GUI - Feature Request

k1ky, 2020-12-16:
Enhancement request to be able to view port configuration(s) and  change the DTD VLAN configuration of ports on AirRouters,  HAP and compatible GLInet  and similar future network switch type devices through the GUI.  Retain changes through the upgrade procedure and make it easy to change the Network Port configuration(s) through the node graphical interface instead of manually editing the swconfig file to support more than one DTD connection.

k1ky, 2022-03-21:
Although not directly related, how about migrating the swconfig file during upgrades? This would streamline our upgrades of MESH nodes that have customized DTD ports set up.

ae6xe, 2022-03-21:
The settings won’t be compatible on all devices after openwrt upgrade.  But maybe from this upgrade forward is an option.

cTurtle98, 2022-12-31:
just wanted to point out that this is similar to #466 and #588

aanon4, 2024-08-18:
With some more modern devices offering configuration (not via swconfig mind) this wont be implemented.

closed 2024-08-18 by aanon4