aredn/aredn #21: Hostnames not propagating (was: Nightly Build 960)

ai6bx, 2020-12-16:
Nightly build 960 seems to possibly have some OLSR issues as nodes names are not consistently resolving to IP addresses and some nodes intermittently drop out as neighbors. I have seen this on Rocket Xw Nanoststion, and PBE 400.

aanon4, 2021-09-09:
Leaving this here to help inform the next person who sees this:

Node will re-synchronize if sequence numbers change dramatically (ie. on a reboot) and it takes them receiving 16 "bad" messages before it does this (these messages are not propagated so there's no storm impact). Very roughly, a node receives a message from another node about once per-second, so a resync should take about 16 seconds. This is happening all the time when nodes are rebooted. Nameserver message are only sent every couple of minutes, but unless your node is in a particular lossy environment, you'd still expect a name to propagate in under 5 minutes max (let's assume the first name update is lost during resync).

Name information received by nodes persists for 30 minutes, so it seems unlikely for this issue to ever be seen, unless it's two nodes being rebooted around the same time (this information does not persist across reboots).

These pieces of information which would be useful to know when this happens:
1. Does the name eventually resolve? After 5 minutes? After 10?
2. After 30 minutes do other nodes start to forget about the node?
3. Does the IP address of the node still work in the network even if the name does not?

ae6xe, 2021-09-09:
1) hostname does not resolve, can stay for days, needs a reboot
2) remote nodes do not know histname
3) node is only reachable by IP address by a direct neighbor (need to confirm)

ae6xe, 2021-09-10:
https://github.com/aredn/aredn/issues/25

Duplicate.

closed 2021-11-07 by ae6xe