aredn/aredn #22: Feature Request: add support for Ubiquiti Unifi Mesh / Unifi Mesh Pro

jmspswny, 2020-12-16:
I've got a couple of Unifi Mesh / Mesh Pros sitting around that I'd love to be able to deploy with this firmware.  They're compelling outdoor installation options.  They're both designed for outdoor deployment, with 802.3af support, and a couple of antenna options.  They're both supported by the current ar71xx release OpenWRT (18.06.2) and additional platform details are available here: https://openwrt.org/toh/ubiquiti/unifiac

I'm happy to help out with with this, just need some direction.

johann8384, 2021-11-02:
How did this work out? Are you still using the Unifi Mesh devices?

What is the preferred version to run on them?

jmspswny, 2021-11-03:
Honestly, I've not tried installing again since the initially reported issues. I'll be taking another crack at this, but not in the next month or so.

cTurtle98, 2022-12-31:
> I'll be taking another crack at this, but not in the next month or so.

its been a year now. any updates?

dman776, 2023-04-19:
is this still a valid request, or, can it be closed?

jmspswny, 2025-09-26:
I just tried a couple of different custom builds, one specifying two `wlan`s, and another not.  In both cases, `arden_init` fails, unable to find the radio(s):

> /usr/bin/lua: /usr/lib/lua/aredn/hardware.lua:75: attempt to index local 'radios' (a nil value)
stack traceback:
        /usr/lib/lua/aredn/hardware.lua:75: in function 'get_radio'
        /usr/lib/lua/aredn/hardware.lua:84: in function 'get_radio_count'
        /usr/local/bin/aredn_init:195: in main chunk
        [C]: ?

@ae6xe - do you recall what the config was that at least had one radio working?

jmspswny, 2025-10-16:
@aanon4 - seems like you're the current best person to ask about this.  I'm happy to do the legwork / build / test images, but I'm not sure where to go digging to start solving the radios not being seen issue.  Thanks.

cTurtle98, 2026-01-04:
fyi the ac mesh comes with an adapter plate in the box that lets you mount it to uisp airmax rocket antennas.
the ac mesh is also an active poe device

[image]

it would be nice to finish support for it so we have a device we can use with rocket antennas that uses active poe