aredn/aredn #25: Hostnames not propagating (was: Nightly Build 960)

ai6bx, 2020-12-16:
Nightly build 960 seems to possibly have some OLSR issues as nodes names are not consistently resolving to IP addresses and some nodes intermittently drop out as neighbors. I have seen this on Rocket Xw Nanoststion, and PBE 400.

aanon4, 2021-09-10:
Here's a working theory:

(First I'm assuming OLSR is running in this failure case because you can still reach the nodes via IP address; if it had failed, the routes would expire and the node would be unreachable. Also, the dtdlink come across just fine, which indicates the name server part is also running).

There is code in the name server part of OLSR which strips out invalid entries when it starts up. Invalid can mean a few things, but one of them is an entry referring to an IP address which doesn't exist on a network interface at that moment. Ignoring mid entries (which are only created if the node name is first seen), we have dtdlink entries with one IP address, and the node name with another (the wifi address). So if the wifi IP was missing at a critical startup point, the entries would never be sent.

This explains the scenario described by @k6ccc. The question then is how can this be triggered? This culling process in the name server part of OLSR happens in one of two ways - either the interval time expires or a name message is received from another node. As the interval timer is 30 seconds, that seems an unlikely source. Receiving a name server message could happen at any time after OLSR starts, assuming at least one interface is active, so that is a possibility. Also, bigger networks mean more name server messages, so more chance this can occur at a bad moment.

The big untested part here is, can OLSR be running when the wifi has no IP address yet? There doesn't seem to be any explicit code requiring it to wait for wifi before starting, but there is code to restart OLSRD when network status changes, and perhaps WiFi getting an IP address is one such change? Without understanding more about the NETIFD process (which manages these things) I cant say yet. Perhaps someone knows more about this bit, or has a good way to reproduce this problem?

ae6xe, 2021-09-10:
As I recall, only dtdlink and tunnel up/down restarts olsr in hotplug, also service advertisement changes.  For timing I did make a change to delay, with no effect:
https://github.com/aredn/aredn_packages/commit/4fe5e8eec06f7900b71a862c645ea374467c7fcd

A manual restart of olsr with a mesh RF link can still exhibit the symptoms.

another duplicate and more info:
https://github.com/aredn/aredn/issues/48

if a node is not in the olsr routing table it will go to default route path, which might reach the node.

aanon4, 2021-09-10:
Is there a good way to reproduce the bug reliability, or failing that a particular hardware setup that will generate the bug with at least some regularity?  I can reproduce the observed failure by starting OLSRD earlier in the boot process (just after S20network) and making sure it doesn't restart when the wifi network comes up. While the symptoms are correct, the way to generate them are a little contrived. One observation though is that the hot plug appears to be invoked when wifi comes up but before the link status is active. Because OLSR checks the link status itself and discards inactive links (even if the network device itself is up) there is a window ... perhaps ... where the wifi might be incorrectly ignored.

I've added "uci set network.wifi.force_link=0" which moves the hot plug event to after the link is active, but without a more realistic way to reproduce I can't say for sure if this is a solution. Has anyone tried this?

ae6xe, 2021-09-10:
I've not found a reliable way to reproduce to date.  force_link = 0 has not been tried that I'm aware of.   It seems to appear more on the hap ac lite device with multiple wlan's, which may give more time for the condition to occur.  However, note that when mesh RF is turned off, there is a eth0 vlan interface that goes no where that substitutes (this approach minimized the changes to be able to turn RF off when not in use).   These symptoms still occur with wifi off.

aanon4, 2021-09-10:
The high vlan device threw me the first time I saw it. I dont think this is wifi specific though (despite me giving that impression), as the link status check is performed on all OLSR devices regardless.

dman776, 2022-01-18:
Can you load 3.22.1.0 and test this again?   There were a few fixes that could have impacted this issue.

closed 2022-04-22 by dman776