aredn/aredn #37: Feature Request: 5GHz MESH on Microtik hAP Lite

k1ky, 2020-12-16:
Formalizing the request for 5GHz MESH operation on the Microtik hAP Lite with the ability to re-task the 2.4GHz radio as Wi-Fi AP operation.  As an additional feature explore the possibility of operating both 2.4GHz and 5 GHz Radios in MESH mode creating a "bridge" to couple devices together on the same network.

cTurtle98, 2022-08-31:
was the issue with this that aredn only operates in 802.11n but part 97
the 5ghz radio is an 802.11ac and there isnt drivers for that for part 97 operations right now

am I correct?

dman776, 2022-08-31:
Correct

cTurtle98, 2022-12-31:
with the new drivers in the latest nightly that adds support for ac devices... is this being re-visited?

ab7pa, 2023-02-08:
This should now be possible on quite a few of the supported radio models: Mikrotik hAP ac lite / ac2 / ac3 as well as the GL.iNet AR750 / 750S.

k1ky, 2023-02-21:
Issue #726 should take place of this as there are now more options and models involved.

closed 2023-02-21 by k1ky