aredn/aredn #38: Feature Request WAN on USB Port - Microtik hAP Lite + others

k1ky, 2020-12-16:
New Feature request for WAN input on the USB port - primarily the new Microtik devices, but also investigate the possibility on UBNT AirRouter and UBNT Rockets with USB Ports.  Cellphones, Portable Hot-Spots, Cellular Mi-Fi devices and USB Cellular Devices have the ability to supply internet via the USB port.  Currently I use a Pepwave Surf On the Go (OTG) device to "bridge" from USB to Ethernet connected to the WAN port of our MESH devices.  I have also used a Cradlepoint router to perform the same task.  Both of those devices have the ability to bridge Wi-Wi Internet and Cellular Internet.  This feature would greatly enhance operations while reducing the equipment needed to carry out the task.

cTurtle98, 2022-12-31:
this would be useful for me as well

k1ky, 2023-01-03:
Another feature (which may require a separate request) would be the ability to use the USB Port as a LAN connection if possible. I power my AR750 (Creta) via the USB port on my Laptop, but have to connect/communicate with the devices either Via Wi-Fi (Mesh AP Mode) or via Ethernet which ties up yet another port on the computer.

cTurtle98, 2023-01-03:
> Another feature (which may require a separate request) would be the ability to use the USB Port as a LAN connection if possible. I power my AR750 (Creta) via the USB port on my Laptop, but have to [...]

Are you talking about the usb a port on the side or the micro USB power input? I dont know if the micro usb power input is hooked up for data or not

k1ky, 2023-05-26:
I'm referring to the USB (A?) port hat is on the side of the HAP units. And the USB(A?) port to the right of the LAN port on GL AR-750 Creta units.