aredn/aredn #4: [Enhancement] Include the HEADING information of the nodes with the coordinates

kp4msr, 2020-12-16:
it will be good to have that information in the systeminfo.json API and in the User Interface of the node, a field to select or write what is the direcction of the node, for example: HEADING NORTH, OR SOUTH EAST, OR EAST, OR WEST... that could help the other users to check which antenna its reachable from his location. .

n2mh, 2020-12-28:
Please give some thought as to what to use in that field for omnidirectional nodes. Not everybody is point-to-point.

kp4msr, 2020-12-28:
It could be always avaliable but optional, not required to fill.

 Or it could be requested only if the antenna type is directional by using a toggle button or switch.

But this is very helful

Sent from my iPhone

> On Dec 28, 2020, at 12:22 PM, Mark <notifications@github.com> wrote:
> 
> Please give some thought as to what to use in that field for omnidirectional nodes. Not everybody is point-to-point.

cTurtle98, 2022-07-31:
would it be useful to have an angle number?
0 is perpendicular to plum, positive is pointed up and negative is pointed down
what do you call that number?
they have it on the ubiquiti design tool

cTurtle98, 2022-07-31:
> Please give some thought as to what to use in that field for omnidirectional nodes. Not everybody is point-to-point.

maybe an antenna beam width field or allow the direction to be like from 90 to 180 for a 90 sector

cTurtle98, 2022-12-31:
I just realised when I commented on this back in july I had not read very closely because my comments are handled here https://github.com/aredn/aredn/issues/4#issuecomment-708711792

aanon4, 2023-11-15:
https://github.com/aredn/aredn/issues/943