aredn/aredn #46: Event Mode - White List - nodes authorized to communicate

ae6xe, 2020-12-16:
Ability to select a list of mesh nodes that may participate in an event.  Traffic from other mesh nodes would be blocked.   Selected on node by node basis for overlapping groups to limit communications.

ab7pa, 2022-07-21:
Seems like the features available in LQM (link quality manager) help to resolve this issue.

VA2XJM, 2022-07-21:
> Seems like the features available in LQM (link quality manager) help to resolve this issue.

You are right. It was my submission on the previous repository. So it can be closed.

dman776, 2022-07-21:
LQM solved the blacklist issue, but, we can close this too if you want

closed 2022-07-21 by dman776

ab7pa, 2022-07-21:
LQM has a **whitelist** feature too, so this has been resolved.  THX

km6lslf, 2023-09-12:
How do we use the whitelist feature with LQM?

ab7pa, 2023-09-12:
Explained in the documentation:
https://docs.arednmesh.org/en/latest/arednGettingStarted/advanced_config.html#link-quality-manager-lqm-settings