aredn/aredn #48: A node will not communicate its name.  Neighbors display the IP address only.

ae6xe, 2020-12-16:
Ref:  https://www.arednmesh.org/content/occasional-dns-resolutionbroadcast-failures-new-release 

After booting, a node does not communicate its name to neighbors.    These neighbors will display an IP address only.    This functionality is part of the OLSR Nameservices plugin.      

If this situation is encountered after the node is booted, then boot again or restart olsrd - "/etc/init.d/olsrd restart".  This action is taken on the node with the IP address showing on its neighbors' mesh status. 

This node can be reached by using the IP address, e.g.  "http://1.2.3.4:8080"

dman776, 2022-01-18:
re-test with 3.22.1.0.   a number of OLSR issues where patched in that release which could affect this issue.

mpannen1979, 2022-01-31:
So far so good with 3.22.1.0+

closed 2022-01-31 by dman776