aredn/aredn #49: Routing locked to gateway even when access to the foreign network or internet is down

ae6xe, 2020-12-16:
When a foreign network is attached to the WAN of a mesh node,  routing will fail when this network is down:

1) when "mesh gateway" is checked, devices across the mesh will continue to attempt to route to this down network.
2) a device on the LAN network of the gateway node will also attempt to route to this down network, unless specifically preventing from accessing, if "disable default route" is checked.   

Since the foreign network is outside the control of the mesh gateway, there is no remedial steps the mesh node can take to resolve, however it would be possible to detect the foreign network is down by a config parameter that defines a pingable IP address or hostname on this network the mesh node could watchdog ping to ensure is available.   

If this address becomes unavailable, the address no longer responds, the node could stop advertising and routing to this foreign network and re-route traffic to the closest default gateway across the mesh network.

WZ0C, 2022-08-09:
Would routing LAN traffic across the mesh to another WAN gateway allow for inadvertent transmission of certain traffic in violation of part 97?  In the current implementation with LAN being routed straight through to WAN for non-mesh traffic, no matter what is sent (OS updates, open web browser tabs, etc.), it doesn't hit the ham bands.

ae6xe, 2022-08-14:
"Would routing LAN traffic across the mesh to another WAN gateway allow for inadvertent transmission of certain traffic in violation of part 97?"

This question is relevant for discussion by the nature that there are internet gateways on the RF network.   This particular issue, if addressed, would not change this discussion.  Rather, fixing this issue would optimize and/or fix some issues for all traffic across RF with the internet.  For example, we continue to have failures with a legitimate part 97 echolink between a repeater and the internet.  Another gateway pops up on the mesh network and suddenly, the repeater echolink doesn't work anymore.

It would be best to make part 97 compliance a separate standalone discussion-issue (maybe already one?).