aredn/aredn #9: Feature Request : API like control/action

xJMV, 2020-12-16:
Hi everyone.

At our association, we use Zabbix to monitor our whole network. This allows for the management team to be able to take certain actions from within Zabbix when a problem occurs. We do not have to share passwords and any team members can stop, start or restart services for example directly from Zabbix dashboard via the «scripts» feature.

It would be interesting to add a «security token/key» parameters in the advanced settings that would be used to auth external apps to make certain action from an HTTP POST request.

Here I think mainly about OLSR restart. I noticed that when OLSR stop working, we receive «No SMNP data collection» alerts. So without having to share passwords, any members could restart OLSR or even automate the process.

Other actions like rebooting the node, activate/deactivate tunnels (in case of RF link maintenance) could also be considered.

NR0X, 2021-12-27:
I second the request. API functionality would be really helpful for automating and controlling multiple nodes.

KM6IAU, 2021-12-27:
+`

cTurtle98, 2022-12-31:
it would be cool to implement support for ansible with this as well

aanon4, 2022-12-31:
https://github.com/kn6plv/ansible4aredn