aredn/aredn #58: Support for MikroTik mANTBox 2 12s

n2mh, 2020-12-28:
There was a discussion on the Forum about this equipment a few months ago

https://www.arednmesh.org/content/mikrotik-rb911g-2hpnd-12s-mantbox-no-channels-1-or-2

And, on a groups.io list, it was indicated that this box has been in use for approximately a year now and is similar to a MikroTik Basebox.

I'd like to order one but I don't want any suprises that there is no official image for it.

73, Mark

ae6xe, 2020-12-28:
Mark,  the forum post confirms this device is working, it just needs an
edit to a file on the node, to define the model string.  I can get that
added into the build shortly.

Joe AE6XE

n2mh, 2020-12-28:
Thanks Joe, appreciate it.