aredn/aredn #103: Unsupported Hardware: Meraki MR16 Trip Report

AbortRetryFail, 2021-05-11:
Meraki MR16s run OpenWRT like a dream, so I was able to get the AREDN mesh firmware built without too much trouble. 

It says to file a ticket with my experience, so here it is: 

Everything seems to work in order except for the regulatory region setting the frequencies for US WiFi. Looks similar to this issue, so I've probably got some configuration missing someplace.
https://www.arednmesh.org/content/mikrotik-rb911g-2hpnd-12s-mantbox-no-channels-1-or-2

Other than that I got 2 of them to link up as expected and pass traffic. Install was super easy from the vanilla OpenWRT LuCi firmware update page.

```
 node: N0CALL-XX-XXX-XXX
model: Meraki MR16

!!!! UNSUPPORTED DEVICE !!!!
boardid: Meraki MR16
Device has not been tested. Please file a ticket with your experiences.

br-lan 00:18:0A:XX:XX:XX
eth0   00:18:0A:XX:XX:XX
eth0.1 00:18:0A:XX:XX:XX
eth0.2 00:18:0A:XX:XX:XX
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  00:18:0A:XX:XX:XX
wlan0-1 02-18-0A-XX-XX-XX-00-44-00-00-00-00-00-00-00-00
wlan1  00:18:0A:XX:XX:XX

/proc/cpuinfo
system type     : Atheros AR7161 rev 2
machine         : Meraki MR16
processor       : 0
cpu model       : MIPS 24Kc V7.4
BogoMIPS        : 452.19
wait instruction    : yes
microsecond timers  : yes
tlb_entries     : 16
extra interrupt vector  : yes
hardware watchpoint : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa         : mips1 mips2 mips32r1 mips32r2
ASEs implemented    : mips16
Options implemented : tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit nan_legacy nan_2008 perf
shadow register sets    : 1
kscratch registers  : 0
package         : 0
core            : 0
VCED exceptions     : not available
VCEI exceptions     : not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='00:18:0a:XX:XX:XX'
hsmmmesh.settings.mac2='XX.XXX.XXX'
hsmmmesh.settings.dtdmac='XX.XXX.XXX'
hsmmmesh.settings.node='N0CALL-XX-XXX-XXX'
hsmmmesh.settings.config='mesh'

```

I do have a license to operate, but I'd rather not dox myself on this Github account. :)

ae6xe, 2021-05-11:
Check the vlan settings to ensure dtdlink and wan interfaces are working.  Prior commits point to where these changes are made, if necessary.

AbortRetryFail, 2021-05-12:
On the AP I've got eth0.1 and eth0.2 interfaces and they get the expected IP addresses. Packet capture from my laptop attached to the AP's Ethernet port shows those on ARP's for the configured WAN IP on VLAN 1 and the mesh traffic on VLAN 2. 

Anything else I should be looking for?

ae6xe, 2021-05-12:
Check the /etc/board.json information.  This is created on firstboot.   Certainly, if bad information, it would need an image update.  However, if OpenWRT supported, this should all be correct (AREDN changes the network definitions and the default is working for this device).    Since this is a dual band device, check to see that when Mesh RF is off that both 2Ghz and 5Ghz are working for LAN AP and/or WAN wifi client.

AbortRetryFail, 2021-05-12:
It looks OK to me.
Here's the contents of /etc/board.json:
```
{
        "model": {
                "id": "mr16",
                "name": "Meraki MR16"
        },
        "led": {
                "wan": {
                        "name": "WAN",
                        "sysfs": "mr16:green:wan",
                        "type": "netdev",
                        "device": "eth0",
                        "mode": "link tx rx"
                },
                "wlan1": {
                        "name": "WLAN1",
                        "sysfs": "mr16:green:wifi1",
                        "trigger": "phy0assoc"
                },
                "wlan2": {
                        "name": "WLAN2",
                        "sysfs": "mr16:green:wifi2",
                        "trigger": "phy0assoc"
                },
                "wlan3": {
                        "name": "WLAN3",
                        "sysfs": "mr16:green:wifi3",
                        "trigger": "phy0assoc"
                },
                "wlan4": {
                        "name": "WLAN4",
                        "sysfs": "mr16:green:wifi4",
                        "trigger": "phy0tpt"
                }
        },
        "network": {
                "lan": {
                        "ifname": "eth0",
                        "protocol": "static"
                },
                "wan": {
                        "ifname": "eth0.1",
                        "protocol": "dhcp"
                },
                "wifi": {
                        "ifname": "wlan0",
                        "protocol": "static"
                },
                "dtdlink": {
                        "ifname": "eth0.2",
                        "protocol": "static"
                }
        }
}
```

Confirming that when I disable the mesh network, I have options for both 2GHz and 5GHz on the LAN AP and WAN WiFi Client. :)

ae6xe, 2021-05-12:
upload the support data file, from link on Administration page at bottom.  let's review this to make sure nothing else is showing up unusual.   If everything is working-testing, then suggest to submit the PR of these changes.

ae6xe, 2021-05-12:
the PR should update the README.md file to add this device in the device table.

dman776, 2021-05-12:
Add changes to the config file to build this image as well.   Joe, is this a new sub target that we need to add to the build process?

Thanks,
Darryl

AbortRetryFail, 2021-05-14:
Like I said before, I'd rather not dox myself with my callsign or MAC addresses of mesh nodes associated with it being posted on Github. If there's a way to upload the support data file privately someplace I'll send it that way. 

In the mean time I'll fork this repo, merge in my changes, and figure out how PRs work. Thanks for the help!

ae6xe, 2021-05-14:
@AbortRetryFail, can send to ae6xe at arrl dot net .
 @dman776 , I've not looked regarding the config settings, we'll see in PR and categorize accordingly, guessing generic.

ae6xe, 2021-05-14:
@AbortRetryFail, 

Please run this command and pass back the output (obscuring in MAC or other details of concern):
iw phy phy1 info

The power settings AREDN is configured for in your changes says max is 30dBm.  However, it appears the flash vendor data is limiting this device on 2GHz to:
			* 2397 MHz [-2] (21.0 dBm)
			* 2402 MHz [-1] (21.0 dBm)
			* 2412 MHz [1] (21.0 dBm)
			* 2417 MHz [2] (21.0 dBm)
			* 2422 MHz [3] (21.0 dBm)
			* 2427 MHz [4] (21.0 dBm)
			* 2432 MHz [5] (21.0 dBm)
			* 2437 MHz [6] (21.0 dBm)
			* 2442 MHz [7] (21.0 dBm)
			* 2447 MHz [8] (21.0 dBm)
			* 2452 MHz [9] (21.0 dBm)
			* 2457 MHz [10] (21.0 dBm)
			* 2462 MHz [11] (21.0 dBm)
			* 2467 MHz [12] (21.0 dBm)
			* 2472 MHz [13] (21.0 dBm)
			* 2484 MHz [14] (21.0 dBm)

Double check, if you try to set it to 30dBm, you'll find it won't go above 21dBm.   If so, the AREDN settings need to know max is 21dBm.  We're not concerned about the 5GHz max, as AREDN firmware does not yet know how to do 2 mesh rf interfaces (or switch between 2 and 5GHz yet).  Thus, no way to set 5GHz to do mesh RF and max power setting isn't an issue for now.

AbortRetryFail, 2021-05-14:
I just tried and you're correct, the power resets back to 21.0 dBm. I'll update perlfunc.pm
Thanks!

AbortRetryFail, 2021-05-14:
Here's the phy1 info for the 5GHz interface:
```
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2261 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * P2P-client
                 * P2P-GO
                 * outside context of a BSS
        Band 2:
                Capabilities: 0x11ce
                        HT20/HT40
                        SM Power Save disabled
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 8 usec (0x06)
                HT TX/RX MCS rate indexes supported: 0-15
                Frequencies:
                        * 5180 MHz [36] (22.0 dBm)
                        * 5200 MHz [40] (22.0 dBm)
                        * 5220 MHz [44] (22.0 dBm)
                        * 5240 MHz [48] (22.0 dBm)
                        * 5260 MHz [52] (22.0 dBm)
                        * 5280 MHz [56] (22.0 dBm)
                        * 5300 MHz [60] (22.0 dBm)
                        * 5320 MHz [64] (22.0 dBm)
                        * 5370 MHz [74] (22.0 dBm)
                        * 5375 MHz [75] (22.0 dBm)
                        * 5380 MHz [76] (22.0 dBm)
                        * 5385 MHz [77] (22.0 dBm)
                        * 5390 MHz [78] (22.0 dBm)
                        * 5395 MHz [79] (22.0 dBm)
                        * 5400 MHz [80] (22.0 dBm)
                        * 5405 MHz [81] (22.0 dBm)
                        * 5410 MHz [82] (22.0 dBm)
                        * 5415 MHz [83] (22.0 dBm)
                        * 5420 MHz [84] (22.0 dBm)
                        * 5425 MHz [85] (22.0 dBm)
                        * 5430 MHz [86] (22.0 dBm)
                        * 5435 MHz [87] (22.0 dBm)
                        * 5440 MHz [88] (22.0 dBm)
                        * 5445 MHz [89] (22.0 dBm)
                        * 5450 MHz [90] (22.0 dBm)
                        * 5455 MHz [91] (22.0 dBm)
                        * 5460 MHz [92] (22.0 dBm)
                        * 5465 MHz [93] (22.0 dBm)
                        * 5470 MHz [94] (22.0 dBm)
                        * 5475 MHz [95] (22.0 dBm)
                        * 5480 MHz [96] (22.0 dBm)
[... 57 lines cut ...]
                        * 5885 MHz [177] (22.0 dBm)
                        * 5890 MHz [178] (22.0 dBm)
                        * 5895 MHz [179] (22.0 dBm)
                        * 5900 MHz [180] (22.0 dBm)
                        * 5905 MHz [181] (22.0 dBm)
                        * 5910 MHz [182] (22.0 dBm)
                        * 5915 MHz [183] (22.0 dBm)
                        * 5920 MHz [184] (22.0 dBm)
        valid interface combinations:
                 * #{ managed } <= 2048, #{ AP } <= 8, #{ P2P-client, P2P-GO } <= 1, #{ IBSS } <= 1,
                   total <= 2048, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Supported extended features:
                * [ RRM ]: RRM
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
```

AbortRetryFail, 2021-05-15:
Okay, sending the PR now after testing a fresh build from my fork. It's worked fine on all 3 nodes I have.
Still toying with the RSSI LEDs, but that's not a critical feature.

commit 9fd900f6 (2021-05-16): Add build target for Meraki MR16 devices. (#104)

closes #103
closes #104
closed 2021-05-16 by ae6xe