aredn/aredn #121: "Mesh RF "design on Basic Setup page

FletchFFletch9, 2021-06-17:
In the Mesh RF section of the Basic Setup page the Enable checkbox would make more sense if it were below the Netmask field. Unchecking the box still allows for the DTD connection, whereas now you would expect that it disables everything because it is above IP address. You might also consider separating the two areas with a blue line. Above the IP address you might call that section DTD and below that Mesh RF. This confusion is due to non-standard checkbox usage. For example, the Gnome design guidelines say, "If toggling a check box affects the sensitivity of other controls, place the check box immediately above or to the left of the controls that it affects. This helps to indicate that the controls are dependent on the state of the check box."

ae6xe, 2021-06-17:
There is work in process to redesign and rewrite the UI, such that any updates to the old UI are independent.   This should be a consideration in the new design regardless.   Fixing in this old UI is unlikely anyone would invest time in.

dman776, 2021-06-17:
Not sure I understand.   This entire section is related to Mesh "RF" (RF not DTD wired).
[image]

FletchFFletch9, 2021-06-17:
If you uncheck Enable and save you still have the IP address and the netmask showing. You can still connect to the node through a DTD connection (at least you can on a hAP ac lite) even though there is no RF signal.  I do this to DTD connect two hAP ac lites hardwired together (one on the roof and one in the EOC two floors down)  where they were also getting a weak RF connection.

FletchFFletch9, 2021-06-17:
[image]

dman776, 2021-06-17:
ah.   gotcha.  Thanks for clarifying.

ae6xe, 2021-06-17:
Mesh RF and DTD are distinct interfaces with different IP addresses and network configurations.     It would be possible to have completely different Mesh RF and DTD sections and independently turn them on/off.    DTD network settings are auto-magically handled behind the scene -- hasn't been a compelling reason to be able to show the settings in the UI and turn on/off to date.  Even though Mesh RF is turned off, the IP address continues to be the primary hostname/IP address of the node, but though the magic of OLSR, the routing figures out to get to the node via the different DTD IP address.