aredn/aredn #122: Checkbox usage on Advanced Configuration page

FletchFFletch9, 2021-06-17:
The checkboxes on the Advanced Configuration page do not follow standard checkbox usage guidelines. When checked the box says ON with a check mark. When unchecked the checkbox says OFF without a checkmark. using the Gnome interface as a design example, it says, "Avoid negative check box labels, as this can be confusing and hard to understand." A checkbox for OFF is a negative label.  Further design guidelines from Windows, "With a check box, users make a decision between two clearly opposite choices. The check box label indicates the selected state, whereas the meaning of the cleared state must be the unambiguous opposite of the selected state. Consequently, check boxes should be used only to toggle an option on or off or to select or deselect an item."

dman776, 2021-06-17:
there are no checkboxes on the Advanced Config page.
[image]

closed 2021-06-17 by dman776

FletchFFletch9, 2021-06-17:
This is on a hAP ac lite, for POE and USB passthrough.

dman776, 2021-06-17:
yes,  my screenshot is from a hAP AC Lite.   The POE and USB passthrough settings are toggles with a slider switch style, not a checkbox.

FletchFFletch9, 2021-06-18:
Sorry, you are right, I though everything had been upgraded to at least 3.21.4.0