aredn/aredn #123: DHCP hierarchical failover

FletchFFletch9, 2021-06-17:
Consider having a delay time option on the LAN DHCP server so there is a hierarchical failover should the primary DHCP server go down. I think this is important for emergency communications. I had this problem working in our EOC where four nodes are tied together and three had the DCHP disabled. The primary node went down and I couldn't connect to the others while in the EOC because none of them were serving IP addresses.

ae6xe, 2021-06-17:
I always leave DHCP on and not merge together the LAN networks of nodes on a switch.   This means every node connected to the switch would have a unique LAN vlan.   Turning DHCP off usually ends up with something not working, because the node is no longer a standard behavior and not reliably interchangeable -- at home, in the park, for an event, etc.  We forget DHCP is off and wondering why things aren't working.   If a DHCP server were to failover to another node, with their LAN networks munged together on the switch, while a node could be reached from the LAN, then the services advertisements would be broken.   There may still be something to fix.  If so, probably best to fix the original failure.

FletchFFletch9, 2021-06-17:
That sounds reasonable. I suppose I could play DCHP roulette and just enable it on all the nodes and if I need a specific service on a specific node I can set a static IP address.

ae6xe, 2021-06-17:
No need to play DHCP roulette if the switch is configured with a unique vlan ID for each node's LAN.  A port on the switch is configured to be on the LAN of only one given mesh node.    The node then has the service advertisement.    If this node fails, then fix this problem, or setup ports on the switch for a different mesh node and setup duplicate/failover service advertisements  --  fault tolerance at the service level.

FletchFFletch9, 2021-06-18:
I'm using a dumb 5-port switch.

ae6xe, 2021-09-10:
Unlikely to change how DHCP standard works, so closing this issue.

closed 2021-09-10 by ae6xe