aredn/aredn #130: bug: snrlog program issues

ab7pa, 2021-06-28:
As identified in this forum post [https://www.arednmesh.org/content/betas-break-charts](url) there is an issue since 3.21.4.0 with the /usr/local/bin/snrlog program which affects the contents of /tmp/snr.dat and the files in /tmp/snrlog/.  Since these files are used to display _Charts_ in the UI, the issue reported by users is that charts for various nodes are missing.  _Realtime_ charts for nodes in the dropdown are generated correctly, but not all RF-linked neighbor nodes appear in the dropdown.  The /tmp/snr.dat and /tmp/snrlog/ files are either missing or zero length.  _Archive_ charts seem to be missing for all nodes.

nc8q, 2021-07-04:
Rollback to previous Nightly Builds and regarding the Charts:Realtime/Archive.
Nightly #359 seems to perform correctly displaying multiple choices by node name in 'Select Device' drop down menu.
(Tested with 2 RF connected devices.)
Around Nightly 368; Select Device shown only 1 choice by node name, yet there were 2 RF connected nodes.
Around Nightly 383; Select Device shows only 1 choice by MAC, yet there were 2 connected nodes.

ab7pa, 2021-07-11:
Closed by Nightly Build 433

closed 2021-07-11 by ab7pa