aredn/aredn #141: Enhancement: Roaming GPS updates

FletchFFletch9, 2021-08-09:
I would like to see a way to POST updated GPS coordinates for mobile nodes. Maybe just an http command with the new coordinates and password for authentication.

slomobile, 2021-12-08:
Location of home does not change because current location of node does.  I would not want to update current  coordinates manually or store password for automatic transmission.  The user entered home GPS coordinates behind the password should probably be maintained as a separate variable from the live roaming coordinates.  

Live values might be updated programmatically by serial, usb, or network NMEA messages.  If that is supported, heading, and velocity might as well be live also.  And they might as well get a json flag.  
Perhaps this could all be included as an example for a custom json and its accompanying custom variables.

dman776, 2022-01-18:
duplicate related to #30

closed 2022-01-18 by dman776