aredn/aredn #147: Graphical display of mesh topology

ggsnpdsn, 2021-09-16:
It seems that it is not supported, how can I see the topology of the mesh?

KM6IAU, 2021-11-06:
Are you suggesting a meshmap running on each node?

Eric KG6WXC is the maker of the [meshmap](https://gitlab.kg6wxc.net/mesh/meshmap) software which can be installed on a RPi.

It would be super cool if it was directly on every node, however I'm uncertain how feasible it would be.

In meshmap's current implementation, there are certain backend requirements.  A RPi is well-suited, but many node radios may not be.  In addition to the backend requirements, there is also a fair amount of traffic generated with every polling... not "tons", but certainly not a negligible amount.

Under normal circumstances--if I understand correctly--nodes don't generally need to poll EVERY other node for OLSR.  The only need to share information with-and-about their neighbors.  Meshmap, on the otherhand, will try to query every node on the network.

Including GPS coords in OLSR would be a great way to achieve this with almost no increase in overhead.  But I'm no expert on how OLSR works.  It may already be there, or it may be impractical for other reasons.  I just don't know yet.

dman776, 2022-03-18:
closing.   no response.

closed 2022-03-18 by dman776