aredn/aredn #156: Have Firmware Update section of Admin page optionally show nightly firmware instead of stable

caseydiers, 2021-10-31:
I'm curious if there is interest in this feature.  If so, I will dive in and work on it.

### Feature  Request:

Currently the software will download and display the stable version of the software.  What I'm proposing is that there is an option in "advanced options" to display nightly builds in the Firmware Update page to simplify the update process for those the typically keep their devices on the nightlys update.

The folder structure for the stable releases (http://downloads.arednmesh.org/firmware/) isn't exactly the same as the nightlys (http://downloads.arednmesh.org/snapshots/trunk/targets/) however, I think it would be possible to:

- change the folder to look in from default to nightly
- dive into the correct folder, for example, from http://downloads.arednmesh.org/snapshots/trunk/targets/ to http://downloads.arednmesh.org/snapshots/trunk/targets/ar71xx/generic/ based on the information located in the hardware type field on the admin page
- locate the correct firmware type based on the sha256sums info located in that folder, or alternately by just parsing the files that are in that folder
- do the rest of the process the same as the default release.

Obviously this would be easier if the build system could generate a .list file for the nightlys as exists in the actual release folder, but it doesn't look like that is necessary to make this work.

If the project was in PHP I'd have contributed already, but I'm ready to pull up my sleeves on a couple of improvements in the UI and learn a bit of perl along the way.  From what I've looked at so far, it isn't too much different from what I'm used to so I should be able to pull it off.

caseydiers, 2021-10-31:
KV3T here btw

KM6IAU, 2021-11-06:
I'd love this.

I just started developing a new webUI.  I share your sentiments about the Perl.  I'd rather it was Python.  The further I get into developing the new webUI, the closer I get to just rewriting it in Python.

I will follow this feature request, and include it in my work.

dman776, 2021-11-06:
Check out the aredn_next repo.  That is the basis for our next generation ui