aredn/aredn #160: Feature request - enable / disable tunnels in bulk

k6ccc, 2021-11-03:
This is a feature request to have the ability to at least be able to select or deselect all tunnels on either the tunnel client or tunnel server page with one click instead of having to select one, wait 10 - 20 seconds, select the next one, wait 10 - 20 seconds, etc.  It gets very tedious if you have a lot of tunnels.  This is more of an issue on the tunnel server page than the tunnel clients page.  A couple options of this:
1)  A "select all" button that would select or deselect all tunnels on the page.
2)  A "Select all and Save" button that would select all tunnels on the page and save.  Would go along with #3 
3)  A "Deselect all and Save" button that would deselect all tunnels on the page and save.
4)  A different "Select all and Save" that would select all tunnels on both client and server page and save.  Would go along with #5 
5)  A different "Deselect all and Save" that would select all tunnels on both client and server page and save. 

On #4 and #5, I'll let you figure out the wording.

I will be the first one to comment that #2 and #4 could be a bit dangerous in that it makes it too easy to enable everything, whereas #1 gives the option to deselect a few before saving.

Purpose of this is mainly a time saver when doing firmware updates - which requires disabling all tunnels and later enabling desired tunnels - one at a time.

k1ky, 2022-03-16:
When doing upgrades, this can be easily achieved by removing the VTUN module from the node, reboot and proceed with upgreade.  Once upgraded, re-install VTUN and all settings are restored without any Checking and Unchecking.

dman776, 2022-03-16:
We may be able to add the service stop command to the update script that stops services during the upgrade.  This would allow us to remove the "disable upload" button code.

closed 2022-03-18 by dman776