aredn/aredn #167: /etc/config/dmz-mode is a non uci formatted file and breaks some uci commands

aanon4, 2021-11-09:
The /etc/config/dmz-mode file is not actually a uci formatted file (on my test system it contains the number 3) and will break some uci commands which act on all contents of /etc/config. For example, if you type 'uci commit' to commit any uci changes which are pending, it will fail and changes will not be saved.

This needs to one migrated to some safer location (and this bug is largely here so I don't forget to do it).

ae6xe, 2021-11-09:
dmz-mode is a legacy implementation from ~2013 or before timeframe, never previously conforming to UCI standard config file.  The value captures 2 things:

A) if value in file is 2 to 5 means the LAN subnet range of 10.x.x.x/29, /28, ... for the usable IP addresses of 5, 13, or 29 counts.
B) if the value (better confirm) of 0 means the LAN is an unrouteable address with a NAT-masq with the rest of the mesh (like a home wifi-router with the internet).   

Likely best place is to move this value into is /etc/config/aredn config file.

aanon4, 2021-11-09:
A few bits of code check for the presence of the file regardless of its value. Looking at how its created, it seems absence of the file is the same as a value of 0 (NAT). I was just going to move the file but I'll move the value into /etc/config/aredn instead.

ae6xe, 2021-11-09:
Note, the master saved settings (from /www/cgi-bin/setup) are in /etc/config.mesh/ , a template of settings, and the run-time /etc/config/ config files are built from the template and hardware specific details -- save to config.mesh, then save settings updates /etc/config (or directly save in both places).   In the future  /etc/config.mesh/ settings could all be moved into an AREDN uci config file or an aredn.json file -- something like openwrt at firstboot reads /etc/board.json of device unique details and creates the /etc/config/ config settings.

ae6xe, 2021-11-09:
"it seems absence of the file is the same as a value of 0 (NAT)" 
In Basic Settings, the firstboot value it defaults to is a LAN of 5 usable IP addresses.

closed 2021-12-15 by dman776