aredn/aredn #192: Status.lua - node count differs

dman776, 2021-12-16:
Nanostation M2
dump_perl2 build
The status and status.lua pages differ in the total and node counts.
status has: 342 and 85  (total/nodes)
status.lua has: 343 and 86  (total/nodes)

aanon4, 2021-12-16:
So they do derive the information in slightly different ways - the lua code asks OLSRD for its routes and then filters them while the perl just looks at the routing table (and applies the same filters).
Is this consistently the case? Quickly looking at a couple of my own nodes and they reports the same numbers both ways. Anything worth noting about this M2 node?

ae6xe, 2021-12-16:
timing difference?  The olsr maintained route tables will have delayed updates, than what olsrd plugins will return live.

dman776, 2021-12-16:
this node is plugged into a hapACLite's DTD port.
i hit refresh on both pages simultaneously.    It is repeatable.   the .lua page counts are always +1

aanon4, 2021-12-16:
Can you post a screen grab of the setup page so I can see the configuration? Not sure why it would be +1 off and my nodes not.

dman776, 2021-12-16:
[image]

dman776, 2021-12-16:
[supportdata-K5DLQ-UNSM2-100-202107291506.tar.gz](https://github.com/aredn/aredn/files/7730511/supportdata-K5DLQ-UNSM2-100-202107291506.tar.gz)

aanon4, 2021-12-20:
Accidentally counting the default route - fixed - https://github.com/aredn/aredn/pull/190/commits/c866d2e519a560b36a4a5ba769dd75f2c9ec2c48

aanon4, 2022-01-20:
Can we close this now?

closed 2022-01-30 by dman776