aredn/aredn #193: Status.lua - unsupported device

dman776, 2021-12-16:
Nanostation M2
dump_perl2 build
The "!!! UNSUPPORTED DEVICE !!!" banner is showing on status.lua (but not on the status page - perl based)

aanon4, 2021-12-16:
Can you send me the output of the following (a lua script):

require('aredn.hardware')
require('luci.util')
=luci.util.dumptable(aredn.hardware.get_radio())

dman776, 2021-12-16:
root@K5DLQ-UNSM2-100:~# lua
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio (double int32)
> require("aredn.hardware")
> require("luci.util")
> =luci.util.dumptable(aredn.hardware.get_radio())
/usr/lib/lua/luci/util.lua:144: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'pairs'
	/usr/lib/lua/luci/util.lua:144: in function </usr/lib/lua/luci/util.lua:140>
	(tail call): ?
	[C]: ?

dman776, 2021-12-16:
Also....
root@K5DLQ-UNSM2-100:/usr/lib/lua/aredn# cat /etc/board.json 
{
	"model": {
		"id": "ubnt,nanostation-m",
		"name": "Ubiquiti Nanostation M"
	},

aanon4, 2021-12-16:
The Ubiquiti stuff has some serious mucking about to get the radio info. How about the output for this:

require('aredn.hardware')
=aredn.hardware.get_board_id()

And this:
/usr/local/bin/get_boardid

aanon4, 2021-12-19:
Pinging @dman776 for more info.

dman776, 2021-12-21:
root@K5DLQ-UNSM2-100:~# lua
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio (double int32)
> require("aredn.hardware")
> =aredn.hardware.get_board_id()
0xe012

root@K5DLQ-UNSM2-100:~# /usr/local/bin/get_boardid 
0xe012

aanon4, 2021-12-22:
Okay. This one has me confused then as both the old and new board id functions return the same value, and if I force my build to return that value (fixing the return value of aredn.hardware.get_board_id to "0xe012") the board isn't marked as unsupported.

Above you included board.json which looks truncated and not valid json - does it really look like that? That file being invalid might have some unexpected side effects.

dman776, 2021-12-22:
That was a partial snippet from board.json

Thanks,
Darryl

aanon4, 2021-12-22:
Could you post the whole thing?

dman776, 2021-12-22:
[board.json.zip](https://github.com/aredn/aredn/files/7763206/board.json.zip)

aanon4, 2021-12-24:
So I bought a M2 node but that didn't show this problem, so I guessed it was actually an old M2 node you had. I couldn't source one of those but managed to locally trade a shiny new M5 for an old M5 and that did show the issue! Turns out these older nodes append an extra character return to the information which I needed to strip:

Fixed here: https://github.com/aredn/aredn/pull/190/commits/7fe537269b075b20b7d50582fb6ba9311787b651

aanon4, 2022-01-20:
Can we close this now?

closed 2022-01-20 by dman776