aredn/aredn #195: Need consistent hardware identification

Orv, 2021-12-22:
While Mikrotik, TP-Link and GL.iNet have a field for "machine" type, indicating the hardware type, Ubiquiti uses a hexadecimal board ID, which needs to be looked up to determine what hardware it is.

What's needed is a consistent field within some downloadable file that accurately identifies the hardware the AREDN software is running on.  This will enable accurate metrics to be generated.

aanon4, 2021-12-22:
Is there some structure to this that would be ideal? For example - manufacturer,devicename,revision (not saying this it the ideal, but perhaps you get the idea)?

Orv, 2021-12-22:
For purposes of metrics, manufacturer, and device ID (name) would be fine.  Revision is probably overkill

Orv, 2022-07-23:
This has been resolved with a nightly build.

closed 2022-07-23 by Orv