aredn/aredn #196: Lat/Lon not showing on Mesh Status page

n2mh, 2022-01-02:
Upgraded to Nightly 739 from various earlier versions - 4 nodes showing this issue
The latitude and longitude are not showing on the Mesh Status page but do show on the main Status page. Reviewing /etc/cgi-bin/status and /etc/cgi-bin/mesh shows a difference in how lat/lon is retrieved.

73

n2mh, 2022-01-02:
Additional info: these nodes are all hAP-Lites.

n2mh, 2022-01-03:
Also affecting a Bullet Titanium on 2GHz.

k6ccc, 2022-01-03:
Confirmed on:
Mikrotik hAP AC lite
Mikrotik RB-LHG-2nD-XL
Mikrotik RB-LHG-5HPnD-XL
Ubiquiti Rocket M3 XM
Ubiquiti Rocket M5 XM
Ubiquiti Rocket M5 XW
GL-iNet AR750
GL-iNet USB150
That's every node type I have...

VA2XJM, 2022-01-03:
It would be interesting to copy/paste your sysinfo.json content so it can be compared and see if the format have changed...

sysinfo.json is accessible via http://YOURNODE:8080/cgi-bin/sysinfo.json

Here is mine on latest release and working perfectly with MeshMap:
`
{
   "lon": "-77.7861",
   "sysinfo": {
     "uptime": "172 days, 19:40:27",
     "loads": [
       0.52000000000000002,
       0.42999999999999999,
       0.40999999999999998
     ]
   },
   "interfaces": [
     {
       "name": "tun51",
       "ip": "172.31.227.14"
     },
     {
       "mac": "B8:69:F4:D7:81:E9",
       "name": "br-lan",
       "ip": "10.208.61.33"
     },
     {
       "name": "wlan0",
       "mac": "B8:69:F4:D6:81:E8"
     },
     {
       "mac": "B8:69:F4:D6:81:E9",
       "name": "wlan1",
       "ip": "10.214.129.233"
     },
     {
       "name": "tun52",
       "ip": "172.31.227.18"
     },
     {
       "name": "tun50",
       "ip": "172.31.227.10"
     },
     {
       "name": "tun53",
       "ip": "172.31.227.26"
     },
     {
       "mac": "B8:69:F4:D7:81:E9",
       "name": "eth1.2",
       "ip": "10.215.129.233"
     },
     {
       "name": "tunl0",
       "mac": "00:00:00:00"
     },
     {
       "name": "eth1.0",
       "mac": "B8:69:F4:D7:81:E9"
     },
     {
       "mac": "B8:69:F4:D6:81:E3",
       "name": "eth0",
       "ip": "192.168.123.77"
     },
     {
       "name": "eth1",
       "mac": "B8:69:F4:D7:81:E9"
     }
   ],
   "api_version": "1.8",
   "meshrf": {
     "ssid": "AREDN-10-v3",
     "channel": "-2",
     "status": "on",
     "freq": "2.397 GHz",
     "chanbw": "10"
   },
   "tunnels": {
     "active_tunnel_count": "4",
     "tunnel_installed": "true"
   },
   "lat": "48.0984",
   "grid_square": "FN18cc",
   "node": "VA2XJM-Gateway",
   "node_details": {
     "description": "Services & Tunnel Server",
     "model": "MikroTik RouterBOARD RB952Ui-5ac2nD",
     "mesh_gateway": "0",
     "board_id": "0x0000",
     "firmware_mfg": "AREDN",
     "firmware_version": "3.21.4.0"
   }
 }
`

k6ccc, 2022-01-03:
<html><body>
<div id="content"><div class="tabs "><nav class="tabs-navigation"><ul class="tabs-menu" role="tablist"><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li></ul></nav></div></div><span class="devtools-tab-line"></span><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"><div class="devtools-separator"></div><input class="searchBox devtools-filterinput" placeholder="Filter JSON"></div><div class="panelContent">

  |  
-- | --
lon | "-117.8527"
sysinfo |  
uptime | "52 days, 3:03:35"
loads |  
0 | 1.64
1 | 1.59
2 | 1.54
interfaces |  
0 |  
name | "tun55"
ip | "172.31.90.98"
1 |  
name | "tun51"
ip | "172.31.90.82"
2 |  
mac | "74:4D:28:21:93:C5"
name | "br-lan"
ip | "10.9.60.81"
3 |  
name | "wlan0"
mac | "74:4D:28:20:93:C4"
4 |  
name | "tun59"
ip | "172.31.90.114"
5 |  
mac | "74:4D:28:20:93:C5"
name | "wlan1"
ip | "10.32.147.197"
6 |  
name | "tun50"
ip | "172.31.90.78"
7 |  
name | "tun60"
ip | "172.31.146.141"
8 |  
mac | "74:4D:28:21:93:C5"
name | "eth1.2"
ip | "10.33.147.197"
9 |  
name | "tunl0"
mac | "00:00:00:00"
10 |  
name | "eth1.0"
mac | "74:4D:28:21:93:C5"
11 |  
mac | "74:4D:28:20:93:BF"
name | "eth0"
ip | "192.168.206.21"
12 |  
name | "eth1"
mac | "74:4D:28:21:93:C5"
api_version | "1.9"
meshrf |  
ssid | "AREDN-10-v3"
channel | "-2"
status | "on"
freq | "2397"
chanbw | "10"
tunnels |  
active_tunnel_count | "5"
tunnel_installed | "true"
lat | "34.1098"
grid_square | "DM14bc"
node | "K6CCC-hAP-at-Home"
node_details |  
description | "This is a Mikrotik hAP at my house that provides tunnels into the AREDN network."
model | "MikroTik RouterBOARD RB952Ui-5ac2nD"
mesh_gateway | "0"
board_id | "0x0000"
firmware_mfg | "AREDN"
firmware_version | "661-f1b9491"

</div></div></div></div>
</body>
</html>

aanon4, 2022-01-03:
I've made a pull request to fix this.

dman776, 2022-01-18:
fixed in 3.22.1.0 release

closed 2022-01-18 by dman776