aredn/aredn #203: NB798 led issue on USB150

ab7pa, 2022-01-18:
The GL.iNet USB-150 only has two leds.  The bottom one indicates power, and the upper one should indicate when the USB150 is associated or meshed with other nodes.  In my recent testing, the upper led remained unlit even when the unit was associated/meshed.

dman776, 2022-01-18:
need to check if we have access to that led

K6AH, 2022-01-19:
My USB150 does exactly the same thing.  Works as it should on 3.22.1.0, not on NB798.

dman776, 2022-01-19:
Ahhh.   Could be the lua based linkled function

Thanks,
Darryl

aanon4, 2022-01-19:
I'm sure. We changed how the LED light was determined in the lua code compared to the older perl code, simply pulling a specific led definition from the board.json info. It looks like this isn't going to work, or at least cause too many regressions, so I'm going to revert that code back to the original (which was just a copy of what Perl did).

ab7pa, 2022-01-30:
NB838 fixed this issue.  USB150 link light now illuminates when linked.  THX

closed 2022-01-30 by ab7pa