aredn/aredn #205: NB811 mesh status layout issue

ab7pa, 2022-01-20:
I noticed on today’s NB811 that the layout of the Mesh Status page is slightly different.  If this was not intentional, then maybe it should be looked at in order to have it correspond with the perl version.
[image]

aanon4, 2022-01-20:
If you add .pl to the url you can get old old perl version of this page. Can you post that for comparison?

ab7pa, 2022-01-21:
mesh.pl for comparison
[image]

ab7pa, 2022-01-30:
This has been fixed in subsequent nightly builds.  THX

closed 2022-01-30 by ab7pa