aredn/aredn #219: Relocate "Allow other to use my WAN" feature to advanced configuration

nathantcampbell, 2022-02-16:
Hi,

I'm a member of the Willamette Valley Mesh Network. We've been growing pretty quick so we constantly have new hams getting into the fold. We've found that often a newer ham will check "allow others to use my WAN" during initial setup/configuration thus gating the internet to the rest of the mesh network which is mostly part 97 based. This obviously represents a problem since most internet connections these days use encryption which obscures the content of the traffic.

I would like to propose the consideration of moving this feature to the "advanced configuration" page where it'll still be available for more exotic setups or in case of emergency but would reduce the likelihood of hams accidentally passing encrypted traffic over part-97 links outside a downright emergency.

It would also be nice to add some language that shows while enabling the feature. Something to the effect of "Warning, please ensure the RF network you're a member of isn't using part 97 frequencies as this option will gate encrypted internet traffic to all mesh devices."

Nathan, AI7NC

nathantcampbell, 2022-03-18:
"added more information to direct people to the help page instead"

Hi Dan, does this mean you're adding some sort of notification into the interface that would notify the user the impact of using this setting?

ab7pa, 2022-08-11:
@nathantcampbell, there is a [current PR](https://github.com/aredn/aredn/pull/476) that should accomplish what you were asking for.  In addition to providing a new option, it moves all of them to the Advanced Configuration page.  Hopefully it will come out in the next nightly build.

nathantcampbell, 2022-08-11:
Awesome, thanks for letting me know! I think it's the sensible thing to do.

Nathan Campbell