aredn/aredn #221: node ntp client operation and refresh

ab7pa, 2022-02-16:
Investigate and possibly improve how a node accesses and refreshes time sync with the NTP server address that's entered on the basic setup page.

ab7pa, 2022-03-19:
Low priority since the current design accomplishes its purpose.

ab7pa, 2022-03-19:
A little more research into this issue:
The command line arguments to **ntpclient** ensure that a single clock set occurs. The -s option implies a count of one request to the ntp server.  Using Wireshark to examine the NTP traffic between a node and the local NTP server, there is only one **ntpclient** request from the node (and one server response) which occurs shortly after the node reboots.  Each node keeps **ntpclient** running continuously, but it doesn’t seem to be making any further requests.  Not sure why we keep **ntpclient** running if it’s only making one request at boot time.

ab7pa, 2022-03-19:
One of the ways this issue manifests itself is when nodes are running an app like MeshChat.  The MeshChat message database uses the node clock time to record messages, and the UI then displays those messages "out of order" when the node clocks are way off.

aanon4, 2022-03-19:
We seem to have the worst of two worlds then - the daemon hangs around while only syncing the time once, and node time drifts because it only syncs on reboot.
Does anyone know the history of these choices, and also why ntpclient rather than ntpd or busybox-ntpd?

ab7pa, 2022-03-21:
Tim @aanon4 has done quite a bit of work on this issue over the weekend.  He has designed a package which could replace **ntpclient** with the already available **busybox-ntpd**.  In addition, he tests the ntp server entered in the UI to be sure it is reachable.  He also searches the node's _services_olsr_ file to find other advertised NTP servers and tests them to verify they are reachable.  If he finds reachable NTP servers then he starts ntpd with the entire list of reachable servers.  If there are no reachable ntp servers in the list then ntpd is not started, which ensures that node resources are not wasted running ntp processes if there are no reachable servers available.  Once this package has been tested further, it should be considered as a candidate to replace the existing ntpclient functionality.

ae6xe, 2022-03-21:
As far back as I recall, NTP was configured this way.    I suspect part of the reasoning was that groups did not want internet connections.  Also there was no core function of the node or any applications at the time that was dependent on the node’s time.   The extra traffic would be deemed inefficient.

ab7pa, 2022-03-21:
Currently **ntpclient** continuously uses node resources without updating the clock, which is also inefficient.  What would you propose?

ae6xe, 2022-03-21:
Only run ntp if the time servers are reachable, and change the option to periodically check back with the servers.

k6ccc, 2022-03-21:
It would make sense to me to set time shortly after bootup, and periodically thereafter.  I don't know how crappy the internal clock is on the various devices used, but especially considering environmental effects (think towertop devices in snow), they could be somewhat crappy.  How crappy would dictate a reasonable refresh rate.  I would think once a day or even once a week likely would be fine.

ab7pa, 2022-03-22:
@aanon4 Tim's new code is here for testing purposes:  https://github.com/kn6plv/meshntpd

k6ccc, 2022-03-23:
One more question.  How does am available NTP server need to be listed in order to be found?  Just looking around at the SoCal network, there are several NTP servers listed.  Almost all are listed differently - mostly simply as text (including mine).

Also, I figured out how to install this, and did so on three of my nodes.  According to my NTP server, all three nodes did query the NTP server shortly after bootup - however all those nodes have my NTP server IP entered on the NTP server field on the Setup page, so I don't know if the MeshNTPD found my server, or simply used the NTP server field on the setup page.  At this point, I also don't know if the nodes will repeat the query on some timetable (it's only been about 20 minutes).  Is there a way to tell what the NTPD is doing?

k6ccc, 2022-03-23:
I get all of that.  How forgiving is the NTP search?  The reason I ask is that as I said, almost every one is listed different.  For example:
NTP server 10.9.60.82
NTP 10.41.175.231
NTP 10.35.163.100
NTP (GPS w/ 1PPS stratum 1)
bulletins-nntp
NTP 10.180.106.94 (GPS 1pps stratum 1)

The bulletins-nntp was the only one that was a link.  The rest were just text.  The link went to:  nntp://kf6nmz.local.mesh:5119/
May require some education to those of us running NTP servers on how to list them so that they can be found.

I don't currently have tcpdump on any of the nodes that I added MeshNTPD, but I can easily enough.

k6ccc, 2022-03-23:
I installed tcpdump on one of the nodes that I previously installed the MeshNTPD.  It is obvious that it is only reaching out to touch the sites that have NTP followed by an IP address about every 17 minute.
[NTP dump.txt](https://github.com/aredn/aredn/files/8336844/NTP.dump.txt)
s

aanon4, 2022-03-23:
This is the expression is uses to find servers which *might* be ntp servers (it then culls this list by trying to talk to the servers):

```grep -i ntp /var/run/services_olsr | sed "s/^http:\/\/\(.*\):.*$/\1/"```

It doesn't care about anything except the case insensitive letters "ntp" appearing on the line. The easiest way to see which ones it select is:

```ps | grep ntpd```

ab7pa, 2022-03-24:
Closed in favor of #318

closed 2022-03-24 by ab7pa