aredn/aredn #223: Problems with Nightly 899

k6ccc, 2022-02-23:
After seeing an issue reported with enabling RF via the lua setup page over on the Ventura county MatterMost, I did an update a GL-iNet USBB150, a GL-iNet AR750, a Mikrotik RB-LHG-5HPnD-XL, and a Mikrotik RB-LHG-2nD-XL from Nightly 887 to Nightly 899.  All were done via RF connections, and all reconnected to RF after the update.  So far so good.
I then connected a computer to the LAN port of the LHG-5HPnD and disabled the RF.  When the page advised that a reboot was needed, I could not reboot the node via the Reboot button.  That would result in a "Bad Gateway" error.  If I went back to the IP of the node, and then the setup page, it again wanted me to reboot.  Tried several times both with setup and setup.pl.  Finally found that I could go to the Administration page and successfully reboot the node.
After the reboot, sure enough, RF was off.  When I tried to turn it back on, I was able to enable RF, but again, any attempt to reboot the node from the setup or setup.pl pages resulted in the "Bad Gateway" error.  Again, I was able to reboot the node from Administration page.

aanon4, 2022-02-23:
Any chance you could attach the output from the support tool page just after you get the 'bad gateway' message?

http://<YOUR NODE NAME HERE>.local.mesh:8080/cgi-bin/supporttool

Alternative, if you log into the offending node, can you send the output of the 'logread' command immediately after getting the 'bad gateway' message.

Thanks

k6ccc, 2022-02-23:
The original reported problem from MatterMost was on both a hAP and a NanoStation M2.  That report from K1KY was that when trying to enable RF, he got a "Configuration NOT saved!".  Additionally it said: idconfig: bad address 'eth1' /usr/bin/lua: /usr/lib/lua/utils.lua:455: bad argument # 1 to 'tonumber' (string expected, got nil) stack traceback: [C]: in function 'tonumber' /usr/lib/lua/aredn/utils.lua:455: in function 'mac_to_ip' /usr/local/bin/node-setup:88: in main chunk [C]: ?
Hopefully I typed that right (it was originally screen capture).

k6ccc, 2022-02-23:
How can I post the support file?  girthub says it does not support that file type.

aanon4, 2022-02-23:
In that case just email it to me (tim.j.wilkinson@gmail.com)

k6ccc, 2022-02-23:
Sent via E-Mail

aanon4, 2022-02-23:
Couldn't reproduce this on my trusty GL-iNet AR150 but I have a hAP somewhere so will try on that.

k6ccc, 2022-02-23:
Unfortunately the only one I can try turning RF off is the LHG-5HPnD since the other are 25 miles away at home and while not there, my only connection to them is via RF.  I can try the other three nodes this evening.

aanon4, 2022-02-24:
hAP: I get "Configuration NOT saved!" ... but I get that on the current release build also, so not sure that's related to the nightly build.
LHG-5HPnD-XL-US: I can disable and re-enable RF without a problem on the current nightly build.

VA2XJM, 2022-02-24:
@aanon4 Try into private/incognito mode... I faced that numerous times due to browser cache...

aanon4, 2022-02-24:
@k6ccc You mention a screen shot earlier - any chance you could attach it? (I think that's possible here)

k6ccc, 2022-02-24:
I guess I can...

[image]

k6ccc, 2022-02-25:
Reboot problem still exists in Nightly 913 - at least on the RB-LHG-5HPnD-XL (only one I can test that on from work.

aanon4, 2022-02-25:
Can you send me that system data dump again? Please do it just after the reboot fails. Thanks.

k6ccc, 2022-02-25:
Here it is...

Jim

https://github.com/aredn/aredn/issues/223#issuecomment-1051145230

k1ky, 2022-02-26:
I also reported that Internet is not passing through nodes updated to the last 3 nightlies, including 913 on HAP & GL AR750.  Interesting to note that VTUN does load, so the nodes have internet access.

aanon4, 2022-02-26:
@k1ky Can you tell me more about how this manifests itself? Does the node itself no longer have access to the Internet (e.g. if you log into the node, what does 'ping 8.8.8.8' report?) Is the node no longer advertising WAN address on the status page? Does this immediately happen after upgrading a node or do you have to do something (perhaps unrelated) before this happens? Does the situation persist across a reboot?

The more information to help me reproduce this the better. Also, if you can send me the output from "http://YOUR_NODE_NAME_HERE.local.mesh:8080/cgi-bin/supporttool" for a failed node that would be great - thanks!

k1ky, 2022-02-26:
@aanon4 - Immediately after upgrading.  There is no WAN listed either way as these nodes are not directly connected to the internet, rather sharing through the gateway node., which is listed.  I can reinstall and send support data if there isn't another Nightly release imminent.

aanon4, 2022-02-26:
@k1ky Thanks for this. Interesting, as the only thing we changed in these builds was switching out a setup page which you'd not touched. The support data would be super helpful (and no new nightly today). I'd still like to understand exactly what "cannot access the internet" means for these nodes, especially as they're reporting a gateway.

k1ky, 2022-02-26:
@aanon4 Further investigation yields that the node does in fact pass internet when the latest nightly 913 is installed.  It's the IperfSpeed modules that break it.  Removing IperfSpeed does not restore it to passing internet.  Support files for each stage forthcoming.

aanon4, 2022-02-26:
Thanks - super helpful (also the system dumps you sent via email). I'm opening a separate bug for the iperfspeed issue

closed 2022-03-11 by dman776