aredn/aredn #233: Installing iperf3speed on Nightly 913 stops the internet working

aanon4, 2022-02-26:
Upgrading to Nightly 913 and then installing the iperf3speed package stops the internet working. Originally reported by @k1ky in https://github.com/aredn/aredn/issues/223

aanon4, 2022-02-26:
Notes:
node-setup is failing to regenerate /etc/dnsmasq.conf after iperfspeed is installed, meaning it no longer hands out a default route to clients but will continue to operate correctly as a node in the network (which doesn't depend on dnsmasq).

k1ky, 2022-02-26:
Note: Confirmed that there are NO issues noted with VTUN installed on N-913.

aanon4, 2022-02-28:
For @k1ky: Incorrect handling of the dmz flag was removing the default route given out by DHCP. This didn't take effect until the node-setup script was first run after the update, which in this case happens as part of the iperfspeed package install process. So, once that install happened, the node no longer handed out a default route to your computer connecting via LAN. If you have a chance to check the next nightly build that would be much appreciated.

closed 2022-03-11 by dman776