aredn/aredn #249: NB942 issues

ab7pa, 2022-03-02:
Basic Settings page:
Active Settings box: changed distance setting and it saves/persists after reboots. Changed the Tx Power from dropdown list and it appears to save, but does not persist after reboot.  Always reverts to the max setting in the list.

ab7pa, 2022-03-02:
Also, on the Ports page: checking the "Do Not Propagate" checkbox and creating a DNS Alias are supposed to display different colored entries on the Mesh Status page.  Here is the original PR for that:
https://github.com/aredn/aredn_ar71xx/pull/569

ab7pa, 2022-03-02:
I also just noticed that all of the saved settings in the Advanced Configuration page are blank.  I had not noticed that happening after a SAVE --> REBOOT before.  Will verify that issue as well.

Yes, just confirmed that Set to Default followed by Save Setting on any row in Adv Config will blank all the rows of that table -- sometimes also resulting in a Bad Gateway message.

aanon4, 2022-03-03:
I opened this last comment as a new bug. Let's not add more to this one.

ab7pa, 2022-03-05:
This issue can be closed.

closed 2022-03-05 by dman776