aredn/aredn #253: I also just noticed that all of the saved settings in the top 16 rows of the Advanced Configuration page are blank.  I had not noticed that happening after a SAVE --> REBOOT before.  Will verify that issue as well.

aanon4, 2022-03-03:
I also just noticed that all of the saved settings in the top 16 rows of the Advanced Configuration page are blank.  I had not noticed that happening after a SAVE --> REBOOT before.  Will verify that issue as well.

Yes, just confirmed that Set to Default followed by Save Setting on any of the upper 16 rows in Adv Config will blank all those rows of the table -- sometimes also resulting in a Bad Gateway message.

_Originally posted by @ab7pa in https://github.com/aredn/aredn/issues/249#issuecomment-1057494325_

aanon4, 2022-03-03:
@ab7pa Can you provide a way to reproduce this. I tried what you seem to describe without any issues:
1. Boot up node.
2. Go to Advanced Configuration Page. All boxes are currently populated with the correct information.
3. Hit "Set to Default" on the top box. Nothing changes.
4. Hit "Save" on top box.
5. Reboot
6. After reboot everything on the Advance Page is still okay.

If you could provide a similar set of steps which makes it fail, that'd be fantastic!

ab7pa, 2022-03-03:
Sent a screen capture video separately showing this behavior after a fresh reboot with NB942.

aanon4, 2022-03-03:
https://github.com/aredn/aredn/pull/255 (in part)

ab7pa, 2022-03-04:
This issue is still reproducible with NB956 this morning.

aanon4, 2022-03-04:
Sigh - but I'm not surprised.
Could I ask a favor - can you send me the system data dump again, but this time right before the finally reboot before you see the issue?

ab7pa, 2022-03-04:
Sent 3 support bundles separately, each at key steps in node life cycle.

ab7pa, 2022-03-05:
This issue can be closed.

closed 2022-03-05 by dman776

dman776, 2022-03-05:
Looks like line 71 is the culprit