aredn/aredn #276: 977 - Advertised Service error

k6ccc, 2022-03-09:
On my hAP (Nightly 977), I happened to be looking at the "Port forwarding..." page and it showed an error on the Advertised Service for the website that runs on an attached RasPi.  Confirmed that the webpage is working correctly, and nothing I did would clear the error.  Finally deleted the advertised service and rebuilt it.  Getting the same error.  I was on the page to attempt to set up a port forward, and could not save it because of this error. 
[image]

I have a downloaded support file, but unable to upload it here.

Jim

aanon4, 2022-03-09:
Could you retry with Nightly 991?

k6ccc, 2022-03-10:
I will attempt to have time this evening and let you know.

Jim

https://github.com/aredn/aredn/issues/276#issuecomment-1063497602

ab7pa, 2022-03-10:
This was fixed in [PR 269](https://github.com/aredn/aredn/pull/269)

k6ccc, 2022-03-10:
Had a few minutes so updated.  The Advertised service was fixed, but the upgrade blew away all my tunnels.

aanon4, 2022-03-10:
That one will be fixed in tomorrow's nightly.

k6ccc, 2022-03-10:
Rebuilt all 13 tunnels manually - fortunately I have a spreadsheet with all the required information.  Two things of note:
1) The Tunnel Server Network IP range changed.
2) The Tunnel Server page does not show the tunnels active, however Mesh Status does show them active, and the far ends (on the could I could easily check) show the tunnels active on their respective Tunnel Client pages.

aanon4, 2022-03-10:
Re: 1 - do you know what it was and what it is now?

k6ccc, 2022-03-11:
Just noticed that with 991, Advertised services are not showing up at all on a Mesh Status - either locally or on a different node.
This was observed on a hAP that has an advertised service of a web page that is running on a RasPi, and also an IP phone.  Noticed when I tried to add something as an advertised service, it did not show up - and neither did the pre-existing web page or IP phone.  Created another advertised service and it also did not appear on that hAP, or another node.  Tried it on my other hAP with 977 and the advertised service shows up fine
[image]

aanon4, 2022-03-11:
Is fixed in the Friday build.

closed 2022-03-11 by dman776