aredn/aredn #283: NB 1005 Tunnel server order scrambled

k6ccc, 2022-03-11:
Noticed this on both hAPs that have tunnel server connections.  The displayed order of tunnel server connections is scrambled.  This would affect the far end since the listed order affects the network IPs.
This appears to only be a display issue.  Using the screen capture included, the first entry is correct, but the second line should really be K6CCC-ROCM5-2-GLENDALE.  That tunnel is correctly working with a network IP of 172.31.90.80.  Additionally, the active indications are on the line that goes along with the network IP rather than the client node name.  In other words, only the Client Name is in the incorrect order.
Example, K6CCC-ROCM5-2 (should be the 2nd one listed) with a network IP of 172.31.90.80 is active, but KB6THO-HAPQ-214.141.74 (which should be the 9th listing with a network IP of 172.31.90.108) is not actually active.
Yes, I have a spreadsheet that lists what is supposed to be listed where...

[image]
0

k6ccc, 2022-03-11:
The passwords go along with the Client name.  In other words, the password listed for a Client name is correct, but the Client name and password are not in the correct order to match the Net IP.

aanon4, 2022-03-11:
Could you add a screen shot of this page but generated using the '.pl' extension on the URL - thanks!

aanon4, 2022-03-12:
@k6ccc just circling back to make sure today's nightly fixed this issue for you.

closed 2022-03-17 by dman776