aredn/aredn #285: NB 1005 Advertised Services showing on other nodes

k6ccc, 2022-03-11:
This may be related to the other issue I reported about "Do Not Propagate" being ignored.  I currently have six nodes at home that each have a separate LAN connection to my router (each appears on the router as a separate VLAN).  This allows me to access each node via it's LAN address from my primary computer.  On each of those nodes, the HostName for the router is "K6CCC-Router-1".  All of those are supposed to be "Do Not Propagate", so the duplicate name shouldn't be a problem - and never has been.
One of those six nodes (K6CCC-hAP-at-Home) has an advertised service for a NTP server.  Now when I look at Mesh Status on any node, each of the six nodes is  showing the NTP server on K6CCC-Router-1.
My assumption is that this is because of what is seen as a duplicated hostname.

aanon4, 2022-03-11:
See https://github.com/aredn/aredn/issues/282

aanon4, 2022-03-12:
@k6ccc And the same here, checking today's nightly fixed this.

closed 2022-03-17 by dman776