aredn/aredn #306: Bulk enable/disable for tunnel entries

ab7pa, 2022-03-19:
Several forum posts requested a possible enhancement to the tunneling UI.  Currently each row in the tables must be enabled or disabled one at a time.  This process becomes tedious when the list of clients or servers is long.  Request an enhancement that would bulk check or uncheck all rows in the table.  One click and then Save Changes, instead of a click for every row before Save Changes, if this is possible.

aanon4, 2022-03-19:
Is this desire driven by the need to disable all the tunnels before an upgrade? That's no longer necessary.

ab7pa, 2022-03-19:
It may originally have been motivated by that, which no longer applies.  I'll close this and if the issue is raised again we can get a more detailed use case for it.  THX

closed 2022-03-19 by ab7pa