aredn/aredn #309: 1056 removes nodes from AREDNMesh.org map

k6ccc, 2022-03-19:
Updated several types of nodes from 1049 to 1056 and uploaded data to AREDN servers.  It uploads alright - but not correctly.  Every node I uploaded, disappeared from the map.  Node types updated were:
Rocket M3 XM
Rocket M5 XW
hAP AC lite
LHG 5nD
LHG 2nD
GL iNet USB150
GL iNet AR750

aanon4, 2022-03-19:
Took a look at the payload 1056 is generating (this whole things was broken in 1049) and it appears correct. I ran It through a json validator and it think's is fine. 

The payload no longer contains a flag indicating tunnel software is installed and the api has been bumped to 1.10 so perhaps this it making the backend croak? @dman776

k1ky, 2022-04-01:
Would it not make sense for the payload to report that Tunnel is installed since we will have a mixture of older firmware  nodes with and without Tunnel installed?  These new Nightly loads do appear to publish on the KX6WXC MeshMap running on our stand-alone servers.

dman776, 2022-04-01:
This original issue was resolved a few days ago.it was not a firmware issue.

dman776, 2022-04-01:
The removal of the tunnel indicator was a temporary fix. It will be added back soon to the maps

closed 2022-04-01 by dman776