aredn/aredn #313: 1070 ATH79 cpe210 v 2 did not load correctly

KB4YFK, 2022-03-21:
would not mesh on -2 5mhz for mesh login (signal strength adequate) after loading 1070
- had to local lan into reload 1070 twice, no joy either time.  
- wifi scan returned all wifi public normally seen at 20mhz, and front welcome page/ status page would not load.  
- admin logged in settings were -2 5mhz and was showing all wifi on scan not just -2 5mhz as expected.
- reloaded  3.22.1 with no concerns.

aanon4, 2022-03-21:
With the device in this failed state, could you send me the support data (link at the bottom of the admin page)? Thanks

KB4YFK, 2022-03-21:
[supportdata-KB4YFK-210-2-5-202203211746.tgz.renamed.zip](https://github.com/aredn/aredn/files/8318104/supportdata-KB4YFK-210-2-5-202203211746.tgz.renamed.zip)

KB4YFK, 2022-03-21:
logged in as superuser/root: ---  note: did not see olsrd file-  see screen capture below

root@KB4YFK-210-2-5:~# /usr/local/bin/node-setup -a mesh
/usr/bin/lua: /usr/lib/lua/aredn/hardware.lua:76: attempt to compare string with number
stack traceback:
        /usr/lib/lua/aredn/hardware.lua:76: in function 'wifi_maxpower'
        /usr/local/bin/node-setup:452: in main chunk
        [C]: ?
root@KB4YFK-210-2-5:~#
root@KB4YFK-210-2-5:/etc/config# ls -l
-rw-r--r--    1 root     root             0 Mar 21 20:22 aliases
-rw-r--r--    1 root     root             0 Mar 21 20:22 aliases.dmz
-rw-r--r--    1 root     root          1236 Mar 21 20:22 aredn
-rw-rw-rw-    1 root     root           234 Mar 21 20:22 dhcp
-rw-rw-rw-    1 root     root            70 Mar 21 20:22 dropbear
-rw-------    1 root     root          3516 Mar 21 20:22 firewall
-rw-r--r--    1 root     root          2644 Mar 21 20:22 network
-rw-r--r--    1 root     root          1679 Mar 21 20:22 network_tun
-rw-r--r--    1 root     root             0 Mar 21 20:22 run-fccid
-rw-r--r--    1 root     root             0 Mar 21 20:22 services
-rw-rw-rw-    1 root     root           776 Mar 21 20:22 snmpd
-rw-rw-rw-    1 root     root           773 Mar 21 20:22 system
-rw-rw-rw-    1 root     root           328 Mar 21 20:22 uhttpd
-rw-r--r--    1 root     root            47 Mar 21 20:22 vtun
root@KB4YFK-210-2-5:/etc/config#

aanon4, 2022-03-21:
Thanks - here's the fix https://github.com/aredn/aredn/pull/314

closed 2022-03-23 by dman776