aredn/aredn #322: Ubiquiti PowerBeam M5 (PBE-M5-620-US) only links at 100 Mbps

compsci282, 2022-03-31:
This device out of the box connects at 1 gigabit speeds without an issue.  

After installing an using ARDEN firmware version 3.22.1.0 the link always drops to 100 megabit.  I'm not sure if this is by design or not.  Forcing the connection to use 1 Gbps on several different switches always disconnected the device with ARDEN firmware installed.

I realize that a 100 Mbps ethernet connection is quite appropriate for a 150+ Mbps wifi device, but it would be nice to have the full hardware capability available.  Obviously this is not a high priority.

K6AH, 2022-03-31:
This has been fixed before.  I had this problem a couple years ago with a PBE-M5-620 on the Elsinore Pk link to San Marcos Pk.

dman776, 2022-04-27:
to turn off the 100mbps port lock on devices, comment out the lines in /etc/rc.local:

------
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.
/etc/init.d/dnsmasq start

BOARDID=$(/usr/local/bin/get_boardid)
case "$BOARDID" in
        0xe885|0xe4e5|0xe6e5)
                /usr/sbin/ethtool -s eth0 speed 100 duplex full autoneg off
                ;;
esac

exit 0
-------------

Try this and report back your results.

dman776, 2022-08-09:
closing.  no reply.

closed 2022-08-09 by dman776