aredn/aredn #330: Node Name not resolving, not connecting, showing Wi-Fi address in list N-1145

k1ky, 2022-04-18:
Nightly 1145 on some 32MB Devices isn't connecting (DTD) and is showing the Wi-Fi addy of the non-connecting node in the node list. I've seen this issue with N-1145 on all nodes and with older versions on other nodes.  This is happening so far with some UBNT AirRouter and UBNT Nanobridge devices. Issue probably popped up since the Services fix.
[image]
[image]
[supportdata-K1KY-ARAR-900-LAVERGNE-202204172113.zip](https://github.com/aredn/aredn/files/8503097/supportdata-K1KY-ARAR-900-LAVERGNE-202204172113.zip)
[supportdata-W4GHD-RKM5XW-109-LAVERGNE-100TWR-345-202204172121.zip](https://github.com/aredn/aredn/files/8503098/supportdata-W4GHD-RKM5XW-109-LAVERGNE-100TWR-345-202204172121.zip)

aanon4, 2022-04-18:
The log file of 900-LAVERGNE indicates that OLSRD errored while trying to talk to the dtdlink network interface, although later parts of it recovered. Did rebooting either of the nodes change anything.

Also - as both these nodes appear to have configured Mesh WiFI, the "Service fix" is not active.

k1ky, 2022-04-18:
Reboots no change.

k1ky, 2022-04-18:
Further review after reboots - Selecting the AR-900 unit from units that are DTD connected at the site will now connect to the GUI, but other nodes on the network away from the site don't show the AR-900 in the node list.  I can access the AR-900 by entering http://10.33.237.33:8080/ directly in my browser on a computer connected to the network away from the site.

k1ky, 2022-04-19:
After a day of stabilization, the AR-900 shows up in the node list, but won't connect away from the MESH site. It shows the "redirecting message" and then fails to connect - screen shot below: C:\Users\k1kyt\My Drive\MESH_HAMNET\SUPPORT_DATA\2022-04-18 22_44_46-k1ky-arar-900-lavergne.local.mesh.png

k1ky, 2022-04-20:
More info.  This seems to happen when the affected node doesn't have a or any good RF path or connection when RF is enabled.  I suspect it has something to do with when DTD connections to such nodes resolve to the Wi-Fi addy instead of the LAN addy of the node.

k1ky, 2022-04-20:
This appears to be limited to 32MB UBNT devices.  I am still able to connect DTD to a PBEM5XW with RF not connected. or RF disabled.

k1ky, 2022-04-27:
This issue still exists with Nightly 1191 as of 4/26/22.

k1ky, 2022-05-09:
Issue seems to be resolved as of Nightly 1235. Both no RF and Marginal RF connection now resolve and connect throughout the network.  This appeared to be mainly an AirRouter issue.  I'll close this issue and revisit if any further problems are discovered.

closed 2022-05-09 by k1ky