aredn/aredn #333: rc_stats: lost availability in Nightly Builds (1145 - April 09, 2022)

nc8q, 2022-04-19:
cat /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/
Not finding rc_stats.
Previously I could view rc_status by pasting the top line in an AREDN device root directory,
press tab to select a MAC, then enter rc_stats
to see what MCS values had been tested and chosen.
Is this feature lost or omitted or TBD?

73, Chuck nc8q

aanon4, 2022-04-19:
This is a kernel feature and that's not been changed at all - so not clearly how this could be different.

aanon4, 2022-04-19:
Had a quick look on one of my devices (current nightly) and this is what I find:

```
./kernel/debug/ieee80211/phy0/netdev:wlan0/stations/c4:ad:34:fc:71:62/rc_stats_csv
./kernel/debug/ieee80211/phy0/netdev:wlan0/stations/c4:ad:34:fc:71:62/rc_stats
./kernel/debug/ieee80211/phy0/netdev:wlan0/stations/08:55:31:49:41:2e/rc_stats_csv
./kernel/debug/ieee80211/phy0/netdev:wlan0/stations/08:55:31:49:41:2e/rc_stats
./kernel/debug/ieee80211/phy0/netdev:wlan0/stations/48:8f:5a:fb:d5:e2/rc_stats_csv
./kernel/debug/ieee80211/phy0/netdev:wlan0/stations/48:8f:5a:fb:d5:e2/rc_stats
./kernel/debug/ieee80211/phy0/netdev:wlan0/stations/0c:80:63:13:31:2e/rc_stats_csv
./kernel/debug/ieee80211/phy0/netdev:wlan0/stations/0c:80:63:13:31:2e/rc_stats
```

nc8q, 2022-04-19:
On 4/19/22 16:32, Tim Wilkinson wrote:
>
> Had a quick look on one of my devices (current nightly) and this is 
> what I find:
>
> |./kernel/debug/ieee80211/phy0/netdev:wlan0/stations/c4:ad:34:fc:71:62/rc_stats_csv 
> ...|

> Message ID: ***@***.***>
Hi, Tim:

Hmmmm.

ieee80211/phy1/netdev:wlan1/stations
              ^            ^
My bad.
I needed to substitute '0' with '1'  in both phy# and wlan# folders.
I'll see if I can delete my github post.
3s, Chuck

closed 2022-04-19 by nc8q