aredn/aredn #335: Feature: "Sideloading" firmware files

r1de, 2022-04-20:
HTTP uploading is not good on marginal networks, any little hiccup and the firmware file that is uploaded to the node is incomplete, yet the node does not know this until it is too late and has already shut down various services in order to save RAM to perform the upgrade. (services like ssh and telnet, olsr, almost _everything_)
This is probably why sometimes it takes _several_ tries to upgrade.

This can make the node completely inaccessible until a powercycle. Which may or may not require a "trip to the hill" and can sometimes not be a feasible solution.

One way around this, is to allow another way to upload the firmware file, that _can_ handle network stalls and can resume the transfer.
SSH/SCP can do this and is already included on each and every AREDN device.

I have old patch files, pre Lua, that added this feature for our local custom firmware.
I can add them here, if, just in case, anyone wants to use them to port it to Lua. :smile:

aanon4, 2022-04-20:
I realize this approach is not for everyone, but for people interested, this is how I manage my "fleet" of test nodes using Ansible, an open source configuration tool. I wrote up a simple example (for Linux folk):

https://github.com/kn6plv/ansible4aredn/tree/master/examples/firmware-update#readme

r1de, 2022-04-21:
Ansible is quite the tool and perfect for this, but I still think it should be an option "on device".
Not everyone wants to run a "centralized admin system" on their network.

It's really just the "resiliency" of SCP that I'm after. On our, sometimes marginal, networks it can make or break if an upgrade is successful or not.
Some tiny FTP server could serve the same purpose and use encryption (I know some are still against that).
_Any_ protocol that can resume a transfer after a timeout or "stall" would be better than http uploading.

r1de, 2022-05-06:
never mind... closing this.
:man_shrugging:

aanon4, 2022-05-06:
@r1de did someone say there was an old Perl patch for this? Send it over if you have it and I’ll take a look.

r1de, 2022-05-07:
@aanon4 All this patch does is check if the firmware file has been uploaded, then it enables the button.
The button then just fires off the upgrade in the "standard" way.
since the "Save Settings" box is checked by default, the "sideload" button always saves the settings.
(at least that's how I was doing it before, feel free to come up with your own solution.)
(also, in the patch, I "flipped the bit" for having tunnels installed, in our use case we _had_ to do that, I'm not sure if that is really needed during a firmware upgrade done in this "sideload" way though)

[WXC-CUSTOM-adminPage.patch.txt](https://github.com/aredn/aredn/files/8644260/WXC-CUSTOM-adminPage.patch.txt)

ab7pa, 2022-05-08:
Seems like the most reliable and efficient answer would be to use **rsync**.  I'm just not sure if we can afford the space on each node in order to include the [rsync/rsyncd packages](https://openwrt.org/packages/pkgdata/rsync).

r1de, 2022-05-08:
ssh is already, always included on each and every aredn device.
no need for anything extra.

Closing this since PR #358 got merged already.

Thanks @aanon4 :+1:

closed 2022-05-08 by r1de