aredn/aredn #340: Fix max ETX

Orv, 2022-04-22:
For discussion -

in /www/cgi-bin/mesh, the max ETX is hard-coded to 50, which is a ridiculously high value.
Even in the SoCal network with ~600 nodes (extending all over), the max ETX displayed at this moment is 21.
But typically that's not even usable.  I'm not able to ping any listed nodes with an ETX greater than about 10.
So why display them?  Recommend either 

1. Change hard coded ETX to maybe 15
2. Alternatively, make it configurable, but with a minimum of maybe 10, so users don't shoot themselves in the foot.

Doing either one of them will save some table space and network traffic, no longer broadcasting routes to nodes that in actuality are neither reachable nor usable.

Dropping the ETX in SoCal to 10 would reduce displayed nodes by 5%, trimming off the least reachable.  That's a worthwhile reduction  in OLSR broadcast traffic, IMO.

aanon4, 2022-04-27:
We could move this setting into advanced and group it with 'lowroute' and 'lowmen' settings which also control how much you see on this screen?

Note that this doesn't effect anything except what is displayed - OLSR will do its own thing. The way to effect that is to reduce the max hop count for broadcasts.

dman776, 2022-04-27:
duplicate of #18

closed 2022-04-27 by dman776