aredn/aredn #352: ssh host key issue

Orv, 2022-05-01:
After updating my Kubuntu Linux workstation to version 22.04, no longer able to ssh to AREDN nodes.   This error is received:

obeach@w6bi-house-pc:~$ ssh -p 2222 root@w6bi-vc-siminorth-5g
Unable to negotiate with 10.14.20.232 port 2222: no matching host key type found. Their offer: ssh-rsa

aanon4, 2022-05-01:
Can you see if the following fixes your issue? I'm assuming this linux update disables ssh-rsa keys on the client.

ssh -oHostKeyAlgorithms=+ssh-rsa -p 2222 root@w6bi-vc-siminorth-5g

Orv, 2022-05-01:
Yes, that works.   I believe they've begun deprecating ssh-rsa keys in the Linux distros.   If we don't update the AREDN code with newer keys, I suspect we'll be getting more and more complaints about this.

ab7pa, 2022-05-05:
_dropbear_ can support **ecdsa** as well as **rsa** keys, but from what I can see our version of dropbearkey only has **rsa** support by default.

ae6xe, 2022-08-30:
Given the work around, do we close this issue?  It's not a defect, rather a future encryption method the firmware does not yet support.   Alternatively, we turn this into an enhancement request at the cost of increasing binary size by about 23 kB:

To compile dropbear with the ''DROPBEAR_ECC'' option in menuconfig if you want ECDSA support and document enable admins to generate a key that is the default support in some linux destros.

From ''package/network/services/dropbear/Config.in'':

config DROPBEAR_ECC bool "Elliptic curve cryptography (ECC)" default n help Enables elliptic curve cryptography (ECC) support in key exchange and public key authentication.
            Key exchange algorithms:
              ecdh-sha2-nistp256
              ecdh-sha2-nistp384
              ecdh-sha2-nistp521

            Public key algorithms:
              ecdsa-sha2-nistp256
              ecdsa-sha2-nistp384
              ecdsa-sha2-nistp521

            Does not generate ECC host keys by default (ECC key exchange will not be used,
            only ECC public key auth).

            Increases binary size by about 23 kB (MIPS).

dman776, 2022-08-30:
I think we should add it the build

Orv, 2022-08-30:
I'd like that, obviously.

dranch, 2022-09-20:
I second this feature enhancement for the Dropbear ssh server in the AREDN firmware.  The included SSH client in modern operating systems has deprecated RSA-style keys for some time now.

closed 2022-11-16 by ae6xe