aredn/aredn #355: Move WAN VLAN id from Adv Config to Setup page

ab7pa, 2022-05-04:
Currently the Advanced Configuration page has a setting to change the WAN VLAN ID.  This setting will _not_ be displayed for any device that uses a switch chip, but it _does_ currently appear on devices that have a dedicated WAN port and therefore do not use VLAN tags for WAN access.
[image]

Suggest moving this setting to the Setup page under the "Advanced WAN **Settings**" column where it could be configured with the other advanced WAN settings. If moved to Setup, it should **only** be displayed when the device must use a VLAN tag for WAN access, which would eliminate confusion on devices that have switches or a dedicated WAN port on which VLAN tags are not supported or required. The selection dropdown should validate that VLAN 2 is not selectable since it is already dedicated for DtD links. There should also be a link to the onboard help (?) and a help section should be added that describes this configuration setting.

[image]

dman776, 2022-05-04:
I think this setting should remain on the adv cfg page because:
It is hardware dependent
The input is not validated and could cause major errors if entered incorrectly
The setup page does not have the logic framework for device dependent hiding

my opinions…

ae6xe, 2022-05-04:
Related but different issue to capture.   the user selectable vlan range should not include single digit numbers.  Possibly, there are some standard guidelines we could research to constrain user selections to .    the openwrt code auto assign vlans, 1, 2, 3 ... .    In the current upgrade branch, I've cleaned up aredn usage to better align.   Today, we have a hack to use vlan '0' on some switches for the LAN, to get things working.  In the upgrade branch to current openwrt releases, the switch is using vlan 3 internally for the LAN, but not visible to end users.

ab7pa, 2022-05-06:
Sounds like this idea may not be viable or practical.  Just trying to think of ways to keep Adv Config from becoming a "junk drawer" by putting features into the UI where similar functionality is located.  Will close this issue for now.

closed 2022-05-06 by ab7pa